martijnvwezel/watermeter-esphome #12: Configuration only sensor proposal.

amzaldua, 2023-09-17:
# Problem/Motivation

> (Why the issue was filed)

## Expected behavior

> (What you expected to happen)

## Actual behavior

> (What actually happened)

## Steps to reproduce

> (How can someone else make/see it happen)

## Proposed changes

Are you considering launching a module sensor for external connection to a standard ESP32 board? I favor this solution because the location where the device is installed is accessible to individuals who are not residents of my home

martijnvwezel, 2023-09-17:
So you want a module sensor for external connection to a standard esp32 board? There is on this board already a standard esp32? Dont get what you want.

amzaldua, 2023-09-17:
Yes, that's correct. I would like a sensor module (just the sensor) to connect to an external ESP32 board. This way, your excellent solution would be compatible with more brands of water meters, and for meters located outside of houses, it would be more discreet and safe.

martijnvwezel, 2023-09-19:
l will do this possibly in the future. Currently, focussing on Matter implementation and after that I can have a look in a special sensor board. Only thing to figure out is how to montage it..

martijnvwezel, 2023-09-24:
What brand of wateermeter do you have?

amzaldua, 2023-09-24:
JANZ JV600
[image]

martijnvwezel, 2023-09-24:
[image]

[image]

one of the design concepts, actually working on a sensor version and small sensor version with USB-C

amzaldua, 2023-09-24:
Awesome! Is the microcontroller included, right? This sensor only needs to be plugged into a USB power supply, I suppose...

marco12en3, 2023-09-25:
It should have 2 ways of connecting, usb or a pinheader. With usb cdc every time the amount of liters changes you send an update over serial. With the pinheader you should be able to power the board and get a rotary encoder output and/or a uart output.

martijnvwezel, 2023-10-08:
A small update, I rejected the board design given above. The following board is going to be produced:
[image]
[image]

I think in 2 weeks they will be available to be shipped!

martijnvwezel, 2023-10-08:
@amzaldua  using the sticky double sided tape you can tape it to your watermeter!

martijnvwezel, 2023-10-09:
https://www.tindie.com/products/muino/3-phase-muino-light-sensor-encoder/?utm_source=mandrill&utm_medium=email&utm_campaign=product_approved

martijnvwezel, 2023-10-19:
@amzaldua  you can order it now from the store!

martijnvwezel, 2023-10-30:
[image]
[image]

martijnvwezel, 2023-10-30:
the sensor arrived!