martijnvwezel/watermeter-esphome #16: ESPHome adoption bricks device. Wrong HA integration data

alx-a, 2023-10-10:
TLDR: ESPHome adoption bricks device. Meter can communicate with HA but is sending suspicious data. Requesting build via docker that developer has tested for consistent performance. ESP integration not needed on my end.

In short, after ESPHome adoption (which includes reupload of config through EH and a build on the EH end) the board bricked with the LED stopping to glow. 
I managed to get the board up with the docker  dev build command from your  readme, updating the yaml config  with info relevant to my wifi.  Board now talks to HA. 

The problem is the sensors are consistently reporting an errant minor .2 to .3 value when the meter water  wheel is absolutely not moving.  The graph should at the least be expected to include 150l use within 5 minutes once a day, on average, with filling of a water container. So 150 liter peak consumption is missing from the graph. 

I have experimented with several repositioning over the analog wheel and taping over to exclude errant reflections. My meter, a retired German analog model, has the wheel on the other side of the ones used in your image examples. As far as I can tell, the circle marking on the board is directly over the analog wheel in my physical install.

I would appreciate if you could provide instructions or a flashable binary that you find works as expected. All I need is a means to communicate with HA, bypassing ESPHome adoption is perfectly fine.

[image]
[image]

_Originally posted by @alx-a in https://github.com/martijnvwezel/watermeter-esphome/issues/8#issuecomment-1742179693_

alx-a, 2023-10-10:
I would propose that the next step is deciding whether I should enable logging on my end, alternatively awaiting instructions on troubleshooting steps or trying a different docker build.

Thanks.

martijnvwezel, 2023-10-10:
I am actually working on a yaml file only configuration. This will help in debugging and is able to show all values. If you would like you can connect that yaml to your system, and flash it.
After that You can save the recorded data and I can look what  goes wrong. It seems your meter needs a bit more tuned calibration process.

alx-a, 2023-10-10:
Sure, lets do it. Since the devil is in the details and it's good to have exact same reference points, what would your instructions to move this forward be?

martijnvwezel, 2023-10-10:
SOrry, forgot to add the yaml, here I added, remove .txt:
[watermeter_yaml.yaml.txt](https://github.com/martijnvwezel/watermeter-esphome/files/12858018/watermeter_yaml.yaml.txt)

my suggestion is install addon influx db, connect ha with that influx(can help with that)

I connect my grafana addon again with influx because of my familiarity with grafana.

martijnvwezel, 2023-10-10:
Again thank you a lot for helping me improve the watermeter, its nice if I can double check which meter you have

alx-a, 2023-10-10:
Thanks for the opportunity to participate, it's a fun concept.

About the meter: The manufacturer went through a merger and I was unable to find more information on the specific model. The meter is marked _SPX spanner pollux gmb_x on the cover. The later model is the R100, but I was unable to identify my model. The blurred out markings are presumed to be individual serial stampings and did not return web hits.

Some other things I might have to exclude later if not resolved in software first

0) Is the board sitting right? The glass cover has a slight curvature  and creates a misplaced reflection in the picture. When I position the board over the wheel, the circle marking is, to the best of my ability, directly over the wheel, with the green led positioned over the middle of the wheel, give or take. 

The placement of the meter is in a fairly dark cellar with no sunlight or reflection and no strong light. The board LED is normally the strongest light source in the whole area. Temperature is stable 20c, humidity under 50. 

1) The power supply is a  reputable 2A USB power supply through a 2 meter cable with a C adapter. I have had no issues running NodeMCU boards from the power setup, but it might be useful later on to measure the cleanliness of the supply and get a USB-C cable without an adapter.  I'm not sure when I'll have time to get out the scope though. Cable should be switched within about 7 days.

I'll follow up with more.
[image]
[image]
[image]

martijnvwezel, 2023-10-19:
Do you have data for me?

alx-a, 2023-10-19:
Could you list the exact steps so that its the right data? I resume I should reflash, so starting with the flashing and finishing with export.

martijnvwezel, 2023-10-22:
We worked hard on implementing a full YAMLfile implementation so we can make it more accessible for the community! It is at the moment in develop!

alx-a, 2023-10-22:
So would you like me to wait, or what next step would you suggest?

martijnvwezel, 2023-10-23:
No please update, the changes are even in master branch now. You can use that, please enable speed modus (using the button in the esphome device -> intergreation/devices -> esphome ->watermeter)

alx-a, 2023-10-24:
It may be most efficient if you list the exact commands so that we have same reference to work with. Including if you need me to enable logging or how you would like to recieve the data. So to start,  as an example

1. git clone https://github.com/martijnvwezel/watermeter-esphome
2. cd watermeter-esphome
? edit yaml config, adding local wifi, or go through esp adoption?
3. docker run --rm --privileged -v  ${PWD}:/config   -it ghcr.io/esphome/esphome run  --device=/dev/ttyACM0 "muino-water-meter-esp32.yaml"

There are several other commands / steps that would be most beneficial as listed by you. Otherwise it is difficult to establish where exactly the problems lie.

martijnvwezel, 2023-10-24:
You should actually get a notification that there is a new update ready, you can install it again from the website https://watermeter.muino.nl

For those steps you told me, please checkout the latest and greatest of this repository `git pull` so you have everything ready to go.

Would actually expect you get a notification that there is an update available

alx-a, 2023-10-25:
If you look back at the original issue, it was not , as you have renamed it, "Support new watermeter PN16", 
but a bricked device after ESP adoption. I resolved it with my own steps, but they are far different form the instructions you are referring to.  No update came up as of today. 

 I look forward to getting replicable instructions so as we can clearly establish:

 that it is correct software exporting data in a format that follows your settings.  A  practical  way to approach this is to reflash your sample device, that matches my series, into a state you are content with, note the steps needed, and provide them all here.

martijnvwezel, 2023-10-25:
**Fixing Adoption Issues for Your Water Meter in Home Assistant**

**Step 0: Prepare for Adoption**
- First, ensure the water meter is disconnected from power.
- In Home Assistant, delete all configurations related to the water meter under the ESPHome integration (remove all configs belonging to the water meter).
- Also, delete the water meter ESPHome device by navigating to 'Settings' -> 'Devices Integration' -> 'ESPHome' -> and selecting your water meter device.

**Step 1: Connect the Water Meter**
- Connect your water meter to your computer using a USB-C cable.

**Step 2: Update the Water Meter Firmware**
- Open a Chromium-based web browser.
- Navigate to the section labeled "Update your water meter with a clean binary." Ensure you download the latest version of the firmware, especially if your water meter model is not officially supported.

**Step 3: Connect the Water Meter to Wi-Fi**
- Follow the instructions to connect the water meter to your Wi-Fi network.

**Step 4: Verify Integration in Home Assistant**
- Wait for a few minutes to allow the device to connect to your Wi-Fi network.
- Check Home Assistant to see if the water meter device appears in the list of devices.
- If it appears, add the water meter device to your Home Assistant system.

alx-a, 2023-10-25:
From here:
https://watermeter.muino.nl/
"Go to your ESPHome board and you will see the following soon 
Go through to setup of ESPHome and copy the secret-key
You see now in ESPHome a connected watermeter !!!"

Does the above imply that ESP Adoption should be completed?  

* It may be beneficial to specify that explicitly in the instructions.

* A consolidation instruction document may be beneficial, as right now multiple reference documents may create confusion.

I went ahead with the ESP adoption since it appears the only way (outside of editing yaml, and that is not available with chromium flashing).

ESP adoption log attached. 

After ESP adoption, the API key generated during the adoption was entered in Home Assistant. I attach resulting screen. I should have a 200 L tank fill in about 2 hours, we can see if it spikes in the log.

[image]
[esp-adoption-log.md](https://github.com/martijnvwezel/watermeter-esphome/files/13167462/esp-adoption-log.md)

alx-a, 2023-10-25:
water_liter_sensor supplying continuous data from reinstall in HA. Does not match meter analog spinning wheel, which is not moving at all  most of the time.
[image]

martijnvwezel, 2023-10-26:
As I already told you, the following setup has an extra sauce for you. Your watermeter is not supported because it does not a moon-type of disk like this:
[image]

However, I think it's quite possible to make it work for your disk:
[image]

As you are already supposed to have more knowledge than an average user, I would like you to install this newer firmware and enable the speed modus of the latest version (I added a button that gives a higher sample rate, thus more data). If you for example filled something with 20-200L and save the data in some way. You could send the data to me and I can alter the algorithm to maintain yours. I wonder how many turns the wheel will do and how much water one turn went through the meter?

The documentation is something that will be worked on soon. I am waiting for some help with that!

Can you find the speed modus of the watermeter (settings->Devices->esphome->click on the watermeter -> you see overview of sensor details and the toggle to enable speed modus, you see immediate an increase in the update rate of the sensor information).

Can you save the data somewhere you can easily download the data, I connected my HA with influx db, to store the data without retention period and stuff

alx-a, 2023-10-26:
I can save the data any way that is convenient for you. My concern is esphome trickery and ha corruption, last time the output log from the docker build via your instructions and esphome was different, perhaps we could try building a perfectly replicable build and then bother with HA adoption.

My suggestion would be starting with a replicable build, decide on a data export that excludes possible HA trickery, and then, once the meter reading matches the wheel,  try HA adoption.

However, i can do influx db in HA if that is most convenient for you. 

Confirmed.Speed mode in HA, when on, increases data display speed and the blinking of the green led.

martijnED, 2023-10-26:
Agree if you can pass me some csv file with data, i can perform some data analytics and try to create a version that fits your meter. Please describe the steps what you did careful. Thank you for you help.

Your adoption problem is solved right or you mean the old instructions dont make semse?

alx-a, 2023-10-26:
> Your adoption problem is solved right or you mean the old instructions dont make semse?

I flashed via chromium. new functionality appeared and seems to affect speed of reading. So that part is working. Otherwise, no,  problem not solved, more work needed,  :)

I will work on influx as well as getting a reference container for an exact known water amount. Hopefully over weekend.  Thanks for the opportunity!

martijnvwezel, 2023-10-30:
No problem! I am actually waiting for your data

martijnED, 2023-11-07:
@alx-a  hey could you provide me with some data, Don't need many liters just a few liters. I want a first look at the data!

alx-a, 2023-11-07:
A bit busy with other things at the moment. I have a dockerized HA instance that is limited as to  influx, would have to install and config separately. Have not had the time to set up additional db or data export. Influx is not yet set up. Would be possible later next week.

If you would like a raw db dump from HA, could be  provided faster, but  obviously more data to shift through.

martijnED, 2023-11-07:
Thats perfect too!

alx-a, 2023-11-08:
Sent an email with attached HA database

martijnvwezel, 2024-01-02:
I had some issues with getting the data out of the HA database So I didn't went well. Can we do this using using a csv file way?

martijnvwezel, 2024-03-22:
Closed during inactivity

closed 2024-03-22 by martijnvwezel