martijnvwezel/watermeter-esphome #22: Power over Ethernet hardware version ?

col-panic, 2023-11-23:
# Problem/Motivation

I am thinking about adding this to the water valve in my cellar. There is thick concrete down there everywhere
which is a problem for a WiFi solution. 

## Expected behavior

-

## Actual behavior

-

## Steps to reproduce

-

## Proposed changes

I would like to encourage a hardware version of this meter that only requires a single LAN connection to
both transfer its data and get the power from.

martijnvwezel, 2024-01-02:
Sorry, for the late response, you can buyild it your-self OR I can help with that. These days some boards are available using a power over ethernet board and the rp2040, using a sensor I have build, you should be able to get it to work.

* https://www.tindie.com/products/muino/3-phase-muino-light-sensor-encoder/
* https://www.tindie.com/products/silicognition/poe-featherwing/ (not my product) OR https://www.hackster.io/wiznet/products/w5100s-evb-pico1?ref=project-8e2495

Most likely there are more products out there that will work with my I2C sensor board that works on 3V3 power

martijnvwezel, 2024-01-09:
I will close this issue. When you want more information please reoppen this issue

closed 2024-01-09 by martijnvwezel