martijnvwezel/watermeter-esphome #23: Error during installation

danielrudolf, 2023-12-28:
Thats my Yaml i want to install:

# These substitutions allow the end user to override certain values
substitutions:
  name: "muino-water-meter"

esphome:
  name: "${name}"
  # Automatically add the mac address to the name
  # so you can use a single firmware for all devices
  name_add_mac_suffix: true
  platformio_options:
    board_build.flash_mode: "dio"

  # This will allow for (future) project identification,
  # configuration and updates.
  project:
    name: esphome.muino-water-meter
    version: "2.0.1"

esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: esp-idf

# triggers reading adc values and iterating algoritm
interval:
  - interval: 100ms
    then:
      - output.turn_on: sensor_power
      - component.update: light_sensor_a_dark

  - interval: 1s
    then:
      - component.update: report_liters
      - component.update: report_liters_rounded
      - logger.log:
          level: INFO
          tag: max_average
          format: "a:%d b:%d c:%d"
          args: [ 'id(max_a)', 'id(max_b)' , 'id(max_c)']
      - logger.log:
          level: INFO
          tag: min_average
          format: "a:%d b:%d c:%d"
          args: [ 'id(min_a)', 'id(min_b)' , 'id(min_c)']

# Toggle switch
switch:
  - platform: template 
    optimistic: true
    id: fastupdate
    name: Speed mode
    icon: "mdi:emoticon-cool-outline"

output:
  - platform: gpio
    pin: 7
    id: led
  - platform: gpio
    pin: 6
    id: sensor_power

sensor:
  - platform: adc
    pin: 2
    name: "light_sensor_a_dark"
    id: light_sensor_a_dark
    raw: true
    internal: true
[... 527 lines cut ...]
      pullup: False
      pulldown: False
    inverted: False
    ignore_strapping_warning: False
    drive_strength: 20.0
  id: light_sensor_c_light
  name: light_sensor_c_light
  raw: True
  internal: True
  on_value: 
    - then: 
        - output.turn_off: 
            id: led
        - component.update: 
            id: phase_coarse
  disabled_by_default: False
  force_update: False
  unit_of_measurement: V
  accuracy_decimals: 2
  device_class: voltage
  state_class: measurement
  attenuation: 0db
  update_interval: 60s

It is the Original Config witth my Details for Wifi

martijnED, 2023-12-28:
What is the error or question?

danielrudolf, 2023-12-29:
This is the error log I get when I try to install into ESPhome in HA.

INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/wassersensor.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

sensor.adc: [source :63]

Pin 2 is used in multiple places.
platform: adc
pin:
number: 2
mode:
input: True
output: False
open_drain: False
pullup: False
pulldown: False
inverted: False
ignore_strapping_warning: False
drive_strength: 20.0
name: light_sensor_a_dark
id: light_sensor_a_dark
raw: True
internal: True
on_value:
- then:
- component.update:
id: light_sensor_b_dark
disabled_by_default: False
force_update: False
unit_of_measurement: V
accuracy_decimals: 2
device_class: voltage
state_class: measurement
attenuation: 0db
update_interval: 60s
sensor.adc: [source :72]

Pin 3 is used in multiple places.
platform: adc
pin:
number: 3
mode:
input: True
output: False
open_drain: False
pullup: False
pulldown: False
inverted: False
ignore_strapping_warning: False
drive_strength: 20.0
id: light_sensor_b_dark
name: light_sensor_b_dark
raw: True
internal: True
on_value:
- then:
- component.update:
id: light_sensor_c_dark
disabled_by_default: False
force_update: False
unit_of_measurement: V
accuracy_decimals: 2
device_class: voltage
state_class: measurement
attenuation: 0db
update_interval: 60s
sensor.adc: [source :82]

Pin 4 is used in multiple places.
platform: adc
pin:
number: 4
mode:
input: True
output: False
open_drain: False
pullup: False
pulldown: False
inverted: False
ignore_strapping_warning: False
drive_strength: 20.0
id: light_sensor_c_dark
name: light_sensor_c_dark
raw: True
internal: True
on_value:
- then:
- output.turn_on:
id: led
- delay: 50ms
- component.update:
id: light_sensor_a_light
disabled_by_default: False
force_update: False
unit_of_measurement: V
accuracy_decimals: 2
device_class: voltage
state_class: measurement
attenuation: 0db
update_interval: 60s
sensor.adc: [source :94]

Pin 2 is used in multiple places.
platform: adc
pin:
number: 2
mode:
input: True
output: False
open_drain: False
pullup: False
pulldown: False
inverted: False
ignore_strapping_warning: False
drive_strength: 20.0
name: light_sensor_a_light
id: light_sensor_a_light
raw: True
internal: True
on_value:
- then:
- component.update:
id: light_sensor_b_light
disabled_by_default: False
force_update: False
unit_of_measurement: V
accuracy_decimals: 2
device_class: voltage
state_class: measurement
attenuation: 0db
update_interval: 60s
sensor.adc: [source :104]

Pin 3 is used in multiple places.
platform: adc
pin:
number: 3
mode:
input: True
output: False
open_drain: False
pullup: False
pulldown: False
inverted: False
ignore_strapping_warning: False
drive_strength: 20.0
id: light_sensor_b_light
name: light_sensor_b_light
raw: True
internal: True
on_value:
- then:
- component.update:
id: light_sensor_c_light
disabled_by_default: False
force_update: False
unit_of_measurement: V
accuracy_decimals: 2
device_class: voltage
state_class: measurement
attenuation: 0db
update_interval: 60s
sensor.adc: [source :114]

Pin 4 is used in multiple places.
platform: adc
pin:
number: 4
mode:
input: True
output: False
open_drain: False
pullup: False
pulldown: False
inverted: False
ignore_strapping_warning: False
drive_strength: 20.0
id: light_sensor_c_light
name: light_sensor_c_light
raw: True
internal: True
on_value:
- then:
- output.turn_off:
id: led
- component.update:
id: phase_coarse
disabled_by_default: False
force_update: False
unit_of_measurement: V
accuracy_decimals: 2
device_class: voltage
state_class: measurement
attenuation: 0db
update_interval: 60s

martijnvwezel, 2024-01-02:
Please use code block in github for code sharing.

martijnvwezel, 2024-01-02:
I am currently working on this problem. The latest version of esphome has new 'features' that are basically compiler warnings. However they return **failed config**. They also did not include a proper way of warning, so give me a moment so I can change the config.

martijnvwezel, 2024-01-02:
Fixed in pull-request #25

martijnvwezel, 2024-01-02:
Thank you for creating a pull request

closed 2024-01-02 by martijnvwezel

danielrudolf, 2024-01-02:
I say thank you for the quick completion! I'll try it tonight! Thanks! I'll buy you a coffee!

martijnvwezel, 2024-01-02:
keep me  posted

danielrudolf, 2024-01-02:
Now its running! :)