martijnvwezel/watermeter-esphome #28: constant reboot

scottbob09, 2024-01-11:
I plugged in my device and it never stays connected long enough to re-flash

I would have expected it to not do that.

Here is the log of what I see when it does boot.

---
15:34:49.477 -> ESP-ROM:esp32c3-api1-20210207
15:34:49.477 -> Build:Feb  7 2021
15:34:49.477 -> rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:34:49.477 -> Saved PC:0x40049a64
15:34:49.477 -> SPIWP:0xee
15:34:49.477 -> mode:QIO, clock div:1
15:34:49.477 -> load:0x3fcd5810,len:0x1684
15:34:49.477 -> ets_loader.c 79 
15:34:50.742 -> ESP-ROM:esp32c3-api1-20210207
15:34:50.742 -> Build:Feb  7 2021
15:34:50.742 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:34:50.742 -> Saved PC:0x40049a64
15:34:50.742 -> SPIWP:0xee
15:34:50.743 -> mode:QIO, clock div:1
15:34:50.743 -> load:0x3fcd5810,len:0x1684
15:34:50.743 -> ets_loader.c 79 
--

Checked all connections on the board and they seem fine.
Tried using buttons to make it do something different.
tried installing new drivers, and removing the existing

Any ideas?

martijnvwezel, 2024-01-12:
Yes, I actually also had the exact same thing before. give me a moment to write down the steps clear

martijnvwezel, 2024-01-12:
Basically, normally the bootbutton works except of a vague situation, but you can force it to go to bootloader.

Typically, the boot button functions correctly, except under certain unclear circumstances. However, you can manually initiate the bootloader mode.

In the image from below, the necessary procedure is illustrated.

[image]

The key step is to set pin 8 (GPIO9) to a low state. This can be achieved by creating a short circuit. You can use a piece of wire, but I often find it more convenient to use tweezers. There are two ways to create this short: either between the ESP32's housing and pin 8, or between pin 8 and pin 9 (GND). After establishing the short, press the reset button. This action should trigger the device to enter bootloader mode. You only need to maintain the short for a second or two after pressing reset.

I added zoomed into the muino board to show the pins more clear.

[image]

martijnvwezel, 2024-01-12:
We can have a call to assist you doing this!

scottbob09, 2024-01-12:
This procedure worked.  It took a few tries, but once it was not rebooting, I was able to re-flash from the website, and then adopt it into esphome.  Now to get 5v to the hand hole near the street to feed it.  ;)

martijnvwezel, 2024-01-12:
I assume you have the version with two buttons, if not all the effort was for nothing and the bootpin was enough

closed 2024-01-12 by martijnvwezel