martijnvwezel/watermeter-esphome #29: Support of a new device?

teras, 2024-01-13:
# Problem/Motivation

I am thinking to buy a new water meter for my house and I'd like to use a solution like yours. But I am lost what kind of water meters should I buy. I don't live in Nederlands and the market here is a bit different.

I know there other, magnetic, solutions to this problem but I liked yours; the optical solution. Which means that it's important the "face" of it. 

To start the discussion, I am going to upload a few faces, to tell me if could be compatible. None of them has a "moon like" pointer; practically they have a pointer instead. Still, all of them claim to be magnetic.

So here it is -- you'll see that the pattern is more or less the same. Do you think meters like this could be compatible?

Thank you!

[image]
[image]
[image]
[image]

martijnvwezel, 2024-01-14:
Thank you for taking the interest!
You can buy `KiWa V200` if you want a just working solution.
I am not sure if they are actually magnetic tbo, but I didn't check!
It could be compatible never tested and saw it is not possible, however you can just buy the sensor and make it work,
or buy the DIY version of the sensor to make some own version of it.
I can always give support or help you make it work on those sensors, but it your project because I dont have a meter like that here.

I can even give advise on making tjatiki if you even need that :)

teras, 2024-01-14:
Unfortunately I don't live in the Netherlands (I live in Greece) and this specific meter is impossible to find. I don't care about magnetic solutions, and that's the main reason why I am considering a project like yours; either to buy a ready one (my preferred solution) or build it on my own. But, it has to be compatible and available here :(

Thank you anyway for your effort to help.

martijnvwezel, 2024-01-14:
https://www.amazon.nl/2019er-watermeter-water-doorstroming-meetnauwkeurigheid/dp/B00XLBH2HE/ref=sr_1_29?__mk_nl_NL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=L76MK6UTBDM8&dib=eyJ2IjoiMSJ9.b1oZ6FUyOenWNMRFAutLL73E7dIv89AA9hwMViYJ6VkxOJ-TK5L9PyvEYr0yTnfZkFdTrWfSwtK75BhHXuQCemSWyW3UuE5sRu7fktJrfzK0EGMNB9-GH5vj2rtMVYJJU2HmIu5pq-M2cYQfiRGjnQ.Wu4CCQrftiezXkRLaAOW_ypEFfcY5FtW1iXG0jMv5EY&dib_tag=se&keywords=water+meter&qid=1705260973&sprefix=water+mete%2Caps%2C84&sr=8-29

teras, 2024-01-14:
It's not that easy :)

[image]

martijnvwezel, 2024-01-14:
The moment I have donated a device like this I can take a look and try to get it to work :)

martijnvwezel, 2024-01-15:
I will close the issue because there is no bug to be solved

closed 2024-01-15 by martijnvwezel

teras, 2024-01-19:
In order to boost the project and help the community, I could buy one and see how can send it to you (are you coming to FOSDEM? :laughing: ) The real question though is, since every one of them has different faces, the solution would always be for a specific meter, right? 

So practically this won't change much, the idea won't be to help me on a specific brand, but how to make a decent solution that can help more people.

martijnED, 2024-01-19:
There is a change I actually go to FOSDEM :)

You can order a watermeter at my house and I could bring it if I would go!