martijnvwezel/watermeter-esphome #39: Watermeter stops

thomaskoopmans, 2024-03-21:
# Problem/Motivation

> meter stops registering new usage after about 12 hours after reboot. 

## Expected behavior

> continues logging

## Actual behavior

> error in debug log stating low light 

## Steps to reproduce

> reboot or reinstall watermeter and wait 

## Proposed changes

> (If you have a proposed change, workaround or fix,
> describe the rationale behind it)

thomaskoopmans, 2024-03-21:
[image]

thomaskoopmans, 2024-03-21:
Two hunches. USB C power cable not ok. 5 meter length causing power drop. 

Or

water condensation on sensors. Watermeter is installed in damp space

martijnED, 2024-03-21:
your picture doesnt work

thomaskoopmans, 2024-03-21:
[image]

martijnED, 2024-03-21:
if you have a very damp space use nail polish or something else to protect the watermeter.
5M is quite long for wifi devices itself.

Do you get from the logs that the logging that it  is continuesly too dark?

thomaskoopmans, 2024-03-21:
After a period of 12-14 hours the error messages are popping in regularly. 
wifi connection is ok enough. Nothing in logs indicating connection issues. 

Only the light Too Dark error 

nail polish on the print itself. As in top and bottom?

martijnED, 2024-03-21:
The led still works, maybe the led is very dimmed?
You could also install the latest version, that only sends liter updates after no water has been used for a minut. To make sure proper working
If the error pops up once in the while too dark not a big issue. If you have multiple behind each other its an issue.

martijnED, 2024-03-21:
But that it stops sending updates doesnt make sense. Can you have a look at the meter itself?

martijnvwezel, 2024-03-22:
Can you see if there is water damage to the meter ?

thomaskoopmans, 2024-03-25:
Meter looks ok.
[image]
[image]

thomaskoopmans, 2024-03-25:
The WiFi connection is decent and stays online. Can easily access and update the meter via esphome 

I have some colored nailpolish available. Will coat the top and bottom excluding the lights and sensors.

thomaskoopmans, 2024-03-25:
Is there a debug flag I can set to flesh out power issues? Nothing in current logs

martijnED, 2024-03-26:
Yes there is a bug flag but your watermeter got damaged:
If you look closer you see near the capacitor and the black chip corrosion, maybe after some cleaning your power issue is resolved or you need to replace the items. I can help you with that!

But first clean the pcb and hopefully it will be enough.

Does your watermeter is in a wet environment?
[image]

If you power the board and after a few minuts you feel on touch the blackchip get hot?

martijnED, 2024-03-26:
To answer the logging part, I do log but only errors I can change that.

How to clean:
1) Get some cleaning alcohol
2) get old toathbrust and dip it a bit in the alcohol
3) go slowly over the given picture and clean it so that the white stuff is removed.
4) Make sure the placement is clean of water and test if it works now without problems
5) get nail polish and polish the given part so that the part is protected from water.
6) I did not test nail polish on the sensors itself and what it does to the sensoring part so I cannot help you yet with it.

martijnED, 2024-03-26:
Can you measure the width you enlarged the whole with, i will order a new batch where I enlarge those two holes a bit.

martijnvwezel, 2024-03-31:
@thomaskoopmans  Can you give me an update?

thomaskoopmans, 2024-04-01:
Current status after an initial clean and update of firmware:
- is running properly since the 29th
- does still log the [W][light_level:331]: Too dark

```
INFO ESPHome 2024.3.0
INFO Reading configuration /root/config/muino-water-meter-dda564.yaml...
INFO Updating https://github.com/martijnvwezel/watermeter-esphome.git@main
INFO Starting log output from 192.168.10.105 using esphome API
INFO Successfully connected to muino-water-meter-dda564 @ 192.168.10.105 in 0.201s
INFO Successful handshake with muino-water-meter-dda564 @ 192.168.10.105 in 0.071s
[09:31:24][I][app:102]: ESPHome version 2024.3.0 compiled on Mar 29 2024, 08:52:57
[09:31:24][I][app:104]: Project esphome.muino-water-meter version 2.0.3
[09:31:34][W][light_level:331]: Too dark
[09:31:34][W][light_level:331]: Too dark
[09:31:34][W][light_level:331]: Too dark
```

martijnED, 2024-04-01:
I think you life in utrecht right? You can come by and i can give it a prober clean?

closed 2024-04-03 by thomaskoopmans