martijnvwezel/watermeter-esphome #41: Itron Actaris Schlumberger

Y0rk-code, 2024-03-30:
Is there a 3d case to make it fit on the Itron Actaris Schlumberger?

martijnED, 2024-03-31:
No, because I do not have my own I cannot design it. You you need help like a 3d model of the water meter?

shuffleznl, 2024-04-10:
Fitted to our Itron Aquadis+ (which seems incredibly similar) with 3 tie-wraps and some double-sided tape. Seems to seat quite security between the 'stand-offs' where the blue cap normally rests on.

martijnED, 2024-04-11:
Do you mind sharing a picture?

shuffleznl, 2024-04-12:
Not entirely sure it's exactly over the right area, initially it measured (seemingly) too high values (or we had some insane water usage :)). Maybe I'll reset the values in HA, or try to recalibrate (if I can figure that out). 
[image]
[image]
[image]

If you're up for it can perhaps scan the unit (photogrammetry) to co-develop a PCB-mount/adapter.

shuffleznl, 2024-04-12:
Some top-down views from other sources:
https://cdn.store-factory.com/www.compteur-energie.com/content/product_11170581b.png?v=1607702126
https://www.metermarket.co.uk/assets/products/FullSizeRemanifoldnder.jpg
https://forum.mysensors.org/uploads/files/1449418299492-aquadis.jpeg

Angled view for good measure
https://www.metermarket.co.uk/assets/products/_ogImage/FullSizeReITRON-AQUADISnder.jpg

Happy to print & test any prototype mount.

martijnvwezel, 2024-04-15:
Very nice looking!
[image]

maybe we can create some 3D print that can click in those 3 slots

shuffleznl, 2024-04-16:
Will try to design a basic (like Tinkercad, or perhaps Plasticity basic) 2D/3D model when I get around to it. Shouldnt be too hard and at least help with alignment (e.g. now reading declining values again, perhaps tiewraps shifted. Or Muino accounts for the rain water accumulation (rofl) ).

martijnED, 2024-04-17:
Hahaha very nice hope you can fix it!

SuperPino64, 2026-10-25:
Any update, I have the same meter.