martijnvwezel/watermeter-esphome #43: Measurement consistency

smollynl, 2024-04-12:
# Problem/Motivation
In HA I notice that also negative water usage is reported, measurement is not consistent.

[image]
[image]

Firmware: 2024.3.1 (Mar 27 2024, 18:02:23)
Hardware: 2.0.3

## Expected behavior
correct measurement

## Actual behavior
see motivation

## Steps to reproduce
no specific steps

## Proposed changes
-

smollynl, 2024-04-14:
Is it possible to do a firmware downgrade?
( the problems seem to have been started after the upgrade to v3.0.0)

martijnvwezel, 2024-04-14:
It can sometimes happen if the meter is not well attached or, that its just on the edge half of the sinus.  It correct itself and homeassistant can handle thios

smollynl, 2024-04-14:
It's attached with screws through the holes!
Measurements have been off since the upgrade and seem to get worse.

martijnED, 2024-04-15:
Please keep in mind that the latest version updates are not instant anymore and come in with a delay. Some users with poor wifi had issues that eifi timeouts made the meter stop reading values for a moment so liters get missed

smollynl, 2024-04-18:
I think I found the issue, the quality of the USB cable. I tried to readout the logfile and detection was not consistent, this decided me to exchange the cable. I will try for couple of days and come back.

Other questions, what is the function of the "Speed mode" control?
Is it possible to make a version where you can select the delay? I do not like the delay in v3.0 and I do not need it as my wifi is good. I had the intention with the device to make an automation that provides a notification after a certain amount of ltrs water have been used, for instance during showering.

martijnED, 2024-04-19:
Maybe I can create a button for instant upload or so!

smollynl, 2024-04-23:
Hoi Martijn, helaas het is weer terug sinds de afgelopen dagen. Raar?

[image]

martijnED, 2024-04-23:
Het maakt voor het bijhouden in HA niet uit. Gewoon noice op het signiaal.

smollynl, 2024-04-27:
Hoi Martijn,

Ik heb gisteren de spanning er even afgehaald. En vanaf dat moment is de noise weer weg. Dit lijkt toch dat er iets niet lekker zit in de software, de noise komt na verloop van tijd. Hier screenshot van afgelopen nacht zonder iets van noise.
[image]

Kun je aangeven wat deze control doet?
[image]

Thx en regards,
Roger

martijnED, 2024-04-27:
Hey Roger,

Dit is niet perse de software maar een design constrain. Mocht er in jou watermeter omgevingslicht zijn moet de muino sensor constant zich calibreren aan zijn omgeving. Soms kan hij niet helemaal lekker gecalibreerd zijn en moet hij gereset worden om weer exacte calibratie te hebben. Dit kan overdag soms voor wat noise geven op de waterstand. Echter is dit wat HA mee overweg kan gaan en heb dit zo ingesteld dat HA de +- waarde niet optelt bij het totaal zodat jou energie-dashboard er geen problemen mee heeft. 

Het kan ook zo zijn dat je net op grenswaarde zit. Met grenswaarde bedoel ik dat iedere sinus verdeelt is in 14 stappen. Daar tussen heeft het geen zin om accurater te weten wat de waterstand is. Mocht je waterstand net tussen zo een stap zitten dan kan je constant omhoog gerekend worden of omlaag. Vaak heb je dan ook last van omgevingslicht.

Ik ben bezig met plan maken van features. Was nu vooral bezig om tweede versie te maken die low-power is.

Lijst van features:
* toestaan dat 0.17l 
* instelbare update rate + mensen indicatie geven wat het betekend
* deze versie lower power maken en optie voor maken in de

martijnED, 2024-04-27:
KIjk vooral even hier, daar leg ik uit hoe het werkt:

https://martijnvwezel.com/blogs/watermeter_werking_basics/

smollynl, 2024-04-27:
Hoi Martijn,

Bedankt voor de toelichting. Ik had een Utility meter integratie toegevoegd met een reset cycle van 1 dag. Daar zag ik het gedrag dat het verbruik uit elkaar ging lopen als eerst en toen viel me ook de 'teruglever' gedrag op in het energy dashboard (zie screenshot eerste post). Heb de Utility meter nu aangepast en Net consumption aangezet, dit zou nu goed moeten gaan. 

Mijn watermeter zit in de meterkast en zou geen/heel weinig last van omgevingslicht moeten hebben, behalve als af en toe de deur open gaat. Ik zal wel eens een zwarte doek over de meter leggen.

Interessante lijst van features, wat bedoel je met de eerste 0.17l?
En wat is die speed mode?

Grtz,
Roger

martijnED, 2024-04-27:
speedmodus is er nu tijdelijk uit, maar komt terug wanneer ik het gefixed heb. Lastige fix, ikzelf geneer de noise wat energy dashbaord  negeert het