martijnvwezel/watermeter-esphome #46: Accuracy debugging

SGXander, 2024-06-03:
Just installed my muino and it seems to be updating in 1L increments instead of the expected 100ml increments and also less than measured with a jug from a tap. Can the internal sensors be debugged in the logger to determine why this is happening?

martijnED, 2024-06-04:
What sensor do you have, and are you aware of the lower update rates to HA, because of the wifi issue with esphome?

SGXander, 2024-06-04:
I have yours from here: https://muino.nl/product/smart-water-meter-reader/?v=796834e7a283 stuck to this meter: https://www.ebay.co.uk/itm/145705013117 

It has reset to 0 at around 00:30 and despite two showers and a glass of water this morning it still reads zero on both entities.
What is the lower update rate issue with ESPHome?

SGXander, 2024-06-05:
it seems to be resetting to 0 all the time, is the entity not supposed to be a reading of total?
[image]

SGXander, 2024-06-09:
I've repositioned the sensor as close to the dial as possible and it seems to be better but it still produces negative readings: 
[image]

SGXander, 2024-06-16:
is it possible to adjust the lamda to not allow for negative values? I am taking manual readings to figure out whether it is reading negative instead of positive or just false negative values.

martijnvwezel, 2024-06-16:
Sorry for not getting at your questions earlier. Can you see the logs of the meter itself, or even change log level. I have a feeling the min max function needs to be tuned for your watetmeter

SGXander, 2024-06-16:
It doesn't report much normally. what is the module name to change log level and I'll set to debug and capture some for you.

Where can I check/set the min max function?

martijnvwezel, 2024-11-10:
I am currently trying to figure out a new method but this seems to work very well

martijnvwezel, 2025-01-12:
Hey!

Long time no see, but I updated the code to have static values for the offset. Will update on that in the future in the wiki
 Chceck out the wiki for helping in debugging

martijnvwezel, 2025-10-03:
The link to the wiki: https://github.com/martijnvwezel/watermeter-esphome/wiki

closed 2025-10-03 by martijnvwezel