martijnvwezel/watermeter-esphome #47: Device Power Consumption

elduds, 2024-06-05:
Hello from Australia! Thank you for the cool solution!

Here our detached houses typically have the water meter on the front verge, often outside and away from any power source. I am considering options to power the device. Has anyone else solved this problem?

Is there any information available on power consumption, and if not does anyone have any input into which parameters or how to estimate the power requirements in order to size an appropriate solution such as solar PV/battery or rechargeable battery?

thanks!

martijnvwezel, 2024-06-16:
I know one person bought a small rechargeable battery with a solar panel. Voltatec or something the brand was called. Someone else used very long ethernet cable with 12V power supply, and on the end he putted a 12V to 5V converter and that still works I assume.

matheuscanela, 2024-06-19:
Potentially use something like this
https://www.aliexpress.com/item/1005006996856939.html?spm=a2g0o.productlist.main.1.30025f68Die4AG&algo_pvid=8ec30022-b3ea-4d59-b6ab-35c51365980c&algo_exp_id=8ec30022-b3ea-4d59-b6ab-35c51365980c-0&pdp_npi=4%40dis%21AUD%2159.32%2127.88%21%21%21283.13%21133.07%21%402101dee017188019211512469eb3a6%2112000038992202634%21sea%21AU%21931089232%21&curPageLogUid=zAcJsy9YD3Jn&utparam-url=scene%3Asearch%7Cquery_from%3A

Just not sure if the battery is enough to run more than a day -  @martijnvwezel thoughts?

martijnvwezel, 2025-01-12:
How is this working? Yes I ship to that country.  Check the wiki or https://watermeter.muino.nl/ for more info on what meters are supported the one on your picture is supported

andrewnicols, 2025-07-06:
I am also in the same boat. Do we know what the power consumption is?

I’m also keen to check that it’s okay outside. Is the USB-C waterproof? I have no cover from rain available.

martijnvwezel, 2025-10-03:
> `Do we know what the power consumption is?`
> 
> In general, water is far from electricity (for good reasons, by the way), so I would really like to find a way to power the system with a battery, but it is not clear to me if that is very realistic or not

Actually in the netherlands most water meters are placed below the electricity box. The WiFi chip consumes around 20mA

closed 2025-10-03 by martijnvwezel