martijnvwezel/watermeter-esphome #48: water meter reports wrong / too low consumption values after some days of operation

thoha1970, 2024-06-13:
Hi Martijn,
we've already had some chats via Tindie & WhatsApp a few month ago regarding this problem.

Problem:
Water meter is working well and reporting correct values of consumption after a power cycle for a few days (approx.1, 2 or 3 days)
Then the problem starts: the device still record all consumptions, but with very low values (1 - 2 liters, sometimes negative values of -1, -2 liters), even if the real consumption is in a range of e.g. 4 - 50 liters).
After a power cycle the water meter again starts to record the proper values until a few days later the problem starts again.

What I've done so far for trouble shooting this problem:

- mechanical readjustement close to 100% centered to the dial wheel => problem not solved
- install a hood (card box) over the water meter, in order to exclude daylight / environmental light as root cause => problem not solved
- install a new 5V power supply => problem not solved
- install a new USB cable => problem not solved
- test with firmware v3.0.0 => problem not solved
- test with firmware v2.0.3 => problem not solved

Wifi connection is stable and all the time in a range of -50 - -70 dB (I've  created an entity for Wifi signal strenght in order to monitor that).

I've activated log level INFO and found following interesting records:

In case the problem is not existing, following min / max average is logged:
[I][max_average:664]: a:142 b:426 c:108
][I][min_average:667]: a:14 b:16 c:10

In case the problem is existing, following min / max average is logged (values no more change, all the time only "10" is logged):
[I][max_average:664]: a:10 b:10 c:10
[I][min_average:667]: a:10 b:10 c:10

Any idea if this is for any help for further trouble shooting and what could be the root cause?

Side note: also quite often "[W][light_level:302]: Too dark" is shown in the log.

Any idea how we could find out the root cause for this problem?

As a workaround I've now created an automation which restart the power meter every night @ 3:00 PM.
Let's wait if under this condition whether the fault still occurs or if the water meter now records over long term valid value's of consumption.

Best regards, Thomas

martijnED, 2024-06-14:
I am already writing a version for your watermeter, amd will come back to you this weekend about it. Need some time I need to adjust for your watermeter a different version of min max function

thoha1970, 2024-06-14:
ok, thank you. Take the time you need, this is nothing with high priority

martijnvwezel, 2024-06-16:
Did you tried to install an older version on your watermeter?
You could try an older version that gives more debug information.
Back in the days I gettered data in a new created HA, for temporary used (gives really alot of data you dont want to save).
and used speedy mode to capture this data. In influx db you can then export it easily to csv file. 

This file I can analyse, and finetune the values. This is quite difficult if you don't understand the compleet algorithm. So don't worry. This is also not an easy task, but I will make some time to do so.

thoha1970, 2024-06-16:
I've currently running already v2.0.3
Shall I go for a even oder version? v2.0.0?

martijnvwezel, 2024-07-06:
Can we discuss the next steps for this issue. I am having some more time to look into this

thoha1970, 2024-07-06:
yes, let me know what you need.
Currently I'm running v2.0.3 (but v2.0.3 still having the same problem over a few days).
Since I reset the device via an automation every night @ 3:00 the readings are much more stable / really less mis-readings.
However: from time to time I still see negative readings ( -1liter, -2 liter) and sometimes the measure do dot represent the real consumption.

martijnvwezel, 2024-07-06:
As I already sugested on the email is the min max function is the cause of the weird behavior. Looking at the raw data would give me a better view on what is wrong

thoha1970, 2024-07-07:
I've installed and configured InfluxDB, but really struggle how to create a query for the data you need.
Database and User is setup proper, but I'm having no clue how to setup the required query ....
Could you provide me some tips / how to's?

thoha1970, 2024-07-07:
ok, seems I'm getting the needed query setup and running in Influx DB.
I'm recording following entities:
sensa
sensa_dark
sensa_sub
sensb
sensb_dark
sensb_sub
sensc
sensc_dark
sensc_sub

Are these the data / entities you need for analyses?

The device is running in speed mode.
I've switched off the automation which restart the device every night at 3:00.

How long I should run these recordings?

martijnED, 2024-07-07:
Do it for a day. Then i will examine the values if i need more info I will make some more debug options. If you need help with making a csv, there is somewhere abutton for that :p not sure how to exactly influx doesnt work anymore on my ha installation

thoha1970, 2024-07-08:
find attached the InfluxDB csv file, recorded from 07.07.24 9:12 until now:
[Uploading 2024-07-08-12-33 Chronograf Data.csv…]()

thoha1970, 2024-07-08:
I've done not device reset this night at 3:00 and you can see, problem is there again since 0:00 tonight:

[image]

[image]

[image]

martijnvwezel, 2024-07-13:
The data provided is not helping me, can you see variables sen_a ... sen_C?

thoha1970, 2024-07-19:
sorry for the late reply, I've not been notified about your comment 2 weeks ago.

I do not have variables / entitities named sen_a - sen_c .
I only have sensa - sensc, and these are already included in the csv file.

Any idea what I have to do?

thoha1970, 2024-08-05:
I'm currently in vacation in Croatia. Will be back in 2 weeks, Greetings, Thomas

col-panic, 2024-11-05:
My sensor does not seem to work correctly at all. The values are far too low, and i don't understand them:

e.g. its being constantly on 12 but not recording and consumption for the energy dashboard

[image]

Is there anywhere a troubleshoot, or what should i do?

martijnED, 2024-11-05:
Try to restart the Muino meter each day, should be something quite simple for that. I assume the calibration settings seem to explode in your setup. I am researching a new option for calibration but is quite complex problem. And I am trying some different methods to improve. The hardest part of the Muino water-meter is the limitation of memory CPU time.

I checked your meter someone in the past with similar meter had same issue as you, but solved it just by restarting each day to solve exploding value.

martijnED, 2024-11-05:
This weekend I finally have time to work on the algorithm again. I am creating a way for people to make it more simple to export  live data so I can analyse it.

col-panic, 2024-11-05:
@martijnED can the meter be restarted programmatically? I mean I wouldn't wanna go down and un/plug all the time ...

martijnED, 2024-11-05:
Yes is possible because it uses esphome you can do that with an automation

martijnvwezel, 2024-11-10:
I didnt work on it this weekend, I am sick so trying next weekend.

thoha1970, 2024-12-22:
Hi Martijn,
would v3.0.3 improve anything on the problem reported in this issue?
Shall I install it and disable the daily restart automation, in order to get  any additional debug information?

Best regards, Thomas

martijnvwezel, 2024-12-22:
Yes, please with new debug function.

thoha1970, 2024-12-22:
I've got during updating to v3.0.3 following warnings:

[image]

[image]

Is this ok or do I have to install something else?

thoha1970, 2024-12-22:
v3.0.3 ist running, consumption measurement seems to work.
Speed & Debug Mode is ON, daily restart automation is OFF.

Let's wait the coming days, if the issue occurs again.
In case it'll occur again: how get I the debug data extracted from the entity "sensor.debug_json"?
Ther data seems to be unformated and not accessible by copy&paste:

[image]

martijnvwezel, 2024-12-22:
They are changing stuff with esphome and esp-idf, so we can ignore it for now. 
if you got to the sensor you would see something like this: 
[image]

martijnvwezel, 2024-12-22:
You don't need  continues debug speed on
Can you click on the sensor, and copy the json data

martijnvwezel, 2024-12-22:
``` json
{
  "liters": 659,
  "phase": 0,
  "last_reported_liters": 659,
  "last_reported_liters__mili": 659000,
  "max_a": 221,
  "max_b": 211,
  "max_c": 278,
  "min_a": 6,
  "min_b": 5,
  "min_c": 5,
  "upper_bound": 1500,
  "lower_bound": 5
}

```

thoha1970, 2024-12-22:
Is this what you need?

{"liters":13,"phase":0,"last_reported_liters":13,"last_reported_liters__mili":13000,"max_a":170,"max_b":413,"max_c":222,"min_a":11,"min_b":20,"min_c":18,"upper_bound":1500,"lower_bound":5}

thoha1970, 2024-12-22:
Just to get you right: I can now switch debug & speed OFF again?
Do I need to switch it on again, in case the issue is ashown?

martijnvwezel, 2024-12-22:
Can you please run speed modus, take 3 liters and show me a graph, will make a screenshot soon

martijnvwezel, 2024-12-22:
The values look inbounce

martijnvwezel, 2024-12-22:
Can you check if 3 liters is 3 liters?

thoha1970, 2024-12-22:
First trial, speed mode ON:
I've take 3 liter but only 2 liter recorded

Second trial, speed mode ON:
I've take 3 liter but 0 liter recorded

thoha1970, 2024-12-22:
last update of "sensor.liters": 8 minutes ago, way before the second trial

martijnvwezel, 2024-12-22:
With speedmodus it should live update continuesly. Which meter did you had again ?

thoha1970, 2024-12-22:
Zenner 8 ZR 100

thoha1970, 2024-12-22:
did a 3rd trial (10:45 - 10:46), taken 3 liters but only 1 liter recorded:

[image]

thoha1970, 2024-12-22:
First trial done @ 10:33

thoha1970, 2024-12-22:
> With speedmodus it should live update continuesly. Which meter did you had again ?

After installing v3.0.3 indeed updates have been live and continuesly, but now measurement seems to be missed.

Shall I power cycle the device and test again?

martijnED, 2024-12-22:
Yes try that, will add adc values too soon

thoha1970, 2024-12-22:
ok, will cut power for a while and then repeating the test with 3 liter and I'll provide the graph from the startup.
Will the first few (2) liters used for calibration, as you did in former versions?

thoha1970, 2024-12-22:
put the device power on @ 10:20, set speed mode ON and taken 3 liters @ 11:22:

[image]

3 liters recorded, switch now speed mode OFF and observe the further recordings.

martijnED, 2024-12-22:
Watermeter.muino.nl wrote some extra docs on the buttons

thoha1970, 2024-12-22:
after a while no more consumption recordings, neither in speed mode or speed mode OFF.
I just right now flushed the toilet (which means 9 liters consumption) => no update of sensor L / sensor mL.
sensor L remains on current value.
sensor mL toggles in a range < 1liter radomy up / down (speed mode ON).
Any ideas about the root cause?
I'll go back to the yaml file running before (my customized v2.0.3 version).

martijnED, 2024-12-22:
What are the debug json data after restart 3 liters and after a while?

martijnED, 2024-12-22:
I am going to buikd static thresholds in the comming weeks

thoha1970, 2024-12-22:
I'll install v3.0.3 again and provide you following information:

1.) debug json after restart & taken 3 liter
2.) debug json a few hours later, in case no more cosumption recording

thoha1970, 2024-12-22:
after restart, debug mode ON, 4 liter taken and 4 liter recorded:

{
"liters":4,
"phase":0,
"last_reported_liters":4,
"last_reported_liters__mili":4000,
"max_a":174,
"max_b":415,
"max_c":238,
"min_a":71,
"min_b":159,
"min_c":72,
"upper_bound":1500,
"lower_bound":5
}

martijnvwezel, 2024-12-22:
Soon a version with the option for hardcoded values will come to help with the stability

thoha1970, 2024-12-25:
attached debug json from today, I've taken in the last 20 minutes > 15 liters, but just 4 liters recorded:

{
"liters":541,
"phase":4,
"last_reported_liters":541,
"last_reported_liters__mili":541666,
"max_a":5,
"max_b":59,
"max_c":5,
"min_a":5,
"min_b":6,
"min_c":5,
"upper_bound":1500,
"lower_bound":5
}

max & min values are much lower than 3 days ago.

let me know once you've implemented the static thresholds.

Do you need anything more, or can I go back to my customized v2.0.3 version & daily reset automation?

martijnED, 2024-12-25:
Yeah its super clear, tge max function kills himself. I should add an extra statement to have minimal diff. Thank you if you restart and tske two liters wahts is again the json? I jave an simple idea to improve the algorithm more robust now

thoha1970, 2024-12-25:
attached debug json after restart and taken 6,5 liters in debug mode:

{"liters":6,
"phase":3,
"last_reported_liters":6,
"last_reported_liters__mili":6500,
"max_a":175,
"max_b":402,
"max_c":247,
"min_a":69,
"min_b":118,
"min_c":66,
"upper_bound":1500,
"lower_bound":5
}

I'll switch back tomorrow to v2.0.3, let me know if you need anything more whilst v3.0.3 is installed

thoha1970, 2025-01-10:
Hi Martijn,

I've also running v3.0.4 since you've released it last week.
Daily reset automation is off, since v3.0.4 is running.
After running the device for a couple of days: Right now I've taken 3 liters whilst debug activated => the device reports 3 liters consumed.
On the first look I would say: still running good an recording all measurement / water consumation.

Attached also the plot of this action:
[image]

How you judge this plot? Seems to be all ok or do you see strange behaviours / strage values?

Attached also the history.csv file recorded:
[Uploading history.csv…]()

thoha1970, 2025-01-11:
thank you for your feedback.
Is the "extra filter" you've mentioned a setting in the YAML file or "hardware"?

thoha1970, 2025-01-11:
Just as a test, I've reset the device and take another 3 liters:

[image]

The plot looks a bit different

thoha1970, 2025-01-11:
[Uploading history_3L after reset.csv…]()

martijnvwezel, 2025-01-12:
That will be just software. I implemented today the manual offset values for the meter so you can play around with that in the latest version. I also solved the issue with esphome. I figured out where the issues where so now the web upload thing should work again :)

martijnvwezel, 2025-01-12:
You need atleast that smooth function . Will work on that as soon as I have time. And will implement a switch for disabling the background light filter. Maybe that is in your environment the reason for those spikes.

thoha1970, 2025-01-13:
I've installed v4.0.4 right now.
I'm struggling a bit, if the manual offset feature will whether improve my installation or not.
Do you recommend that I should change the offset?
To be honest: I'm having no clue which offset I have to set, based on the data in the debug_JSON.

Rather more I guess, the mentioned smooth filter might help in order to eliminate these spikes.
What's your opinion?

[8 more replies]