martijnvwezel/watermeter-esphome #52: Changes in yaml config file in release 3.0.0 and 3.0.1 don't seem to work

ewoutdenheijer, 2024-08-05:
# Problem/Motivation

Everything work great with the yaml config from release 2.0.4. In 3.0.0 there are quite some (undocumented) changes and this causes thing to not work with my meter (Elster)
Also, the yaml file included in release 3.0.1 has version number 2.0.3, while the one included in release 2.0.4 has version number 2.0.4.

## Expected behavior

Things to work as before

## Actual behavior

A lot of too dark warnings. calibration not working most of the time.

## Steps to reproduce

-

## Proposed changes

roll back to 2.0.4

martijnED, 2024-08-05:
Would you mind sharing a picture of your setup, please remove ids belonging to  your meter.

ewoutdenheijer, 2024-08-05:
Here you are. The fitting appears fine.
[image]

martijnvwezel, 2024-08-06:
https://github.com/martijnvwezel/watermeter-esphome/releases
Here are the releases and what happend

martijnvwezel, 2024-08-06:
What doesn't work for your meter.
The biggest change I did only update the liters to HA after 1 minut of no water usaged change. To solve an issue with some meters that have bad WiFi reception