martijnvwezel/watermeter-esphome #53: Not working with Diehl Metering Watermeter

col-panic, 2024-11-03:
# Problem/Motivation

Trying to set up with a Diel Metering Watermeter, HomeAssistant does not receive any values, communication
however seems to generally work as the device could be adopted.

[image]

## Expected behavior

After 2 litres, the data should be coming in!

## Actual behavior

No incoming usage values. The only value that gets communicated to Home.Assistant after plugging in  is "Speed mode deactivated" (whatever that means, couldn't find a description for speed mode in the documentation)

[image]

## Steps to reproduce

I guess have the same meter as me :)

## Proposed changes

I don't know what is necessary to get it working!

col-panic, 2024-11-03:
After covering the whole setup with a blanket, the sensor seems to work now! So it seems to be a problem with the light! I'm however not sure if the values can be correct:

[image]

How could it possibly decrease?

martijnvwezel, 2024-11-10:
The reason it can decrease is that some watermeters the continues calibration gets out of bound. Some watermeters requeres a daily reset using Home Assistant automation so that the calibrations keeps being stable.

martijnvwezel, 2025-01-12:
I implmented in the latest version a way to make the calibration values static,  watch the wiki the comming days and will make a better explainaton their :
https://github.com/martijnvwezel/watermeter-esphome/pull/61

here the change explaination and how to calculate the values.

martijnvwezel, 2025-01-26:
manual_offset_a/b/c = (max_a/b/c + min_a/b/c)/2
but this is based on the max you look at the graph, not using sensor values of the max and min.