martijnvwezel/watermeter-esphome #54: connectivity issues

Wiebrenk, 2024-11-12:
Couple of items (2 issues, 1 feature request)
1. The connection of the logger is lost every 10min, this gives the following log in HA; 

"Logger: aioesphomeapi.connection
Source: runner.py:189
First occurred: November 11, 2024 at 08:42:33 (144 occurrences)
Last logged: 08:32:30

muino-water-meter-e5b248 @ 192.168.1.204: Connection error occurred: [Errno 104] Connection reset by peer"

2. The setup link (https://watermeter.muino.nl/) provided with the logger does not show the all the installation steps required, as per stated Github installation steps. Its missing the part how to connect to the logger (IP adres and adoption).

3. Main reason i bought the logger is detect leaks (continuous water usage), but the logger only reports when the usage has stopped. Is there an event limit where the water usage is still ongoing but the memory buffer is full and sents out the usage or so?

martijnvwezel, 2024-11-12:
1. Not sure, didnt see this issue before
2. Can you eleberate a  bit more on this, voor mijn beeldvorming 

3. Actually, did not though about this last time I created a fix. I think I need to add some timeout after x time always update. Because it tries to detect if you

Wiebrenk, 2024-11-12:
For item 2: In the package received, there is a installation quide (QR code). If you scan this code you will be diverted to https://watermeter.muino.nl/.  After step 3 of the installation steps, it states; **_"Go in Home assistant to your devices and adopt the Watermeter"._** Which is impossible as the logger is not yet connected to the user WIFI network. 

Looking at the installation steps of this Github, there are extra steps mentioned after step 3 in order to have the unit connected to the user WIFI network;

_**"On the device connected to the Muino Smart Water Meter, go to http://192.168.4.1/ and select your prefered WiFi SSID to connect the Muino Smart Water Meter with and enter the SSID passcode."**_ (also note the spelling error on=once?)

martijnvwezel, 2024-11-12:
I just checked my code, but each minut it always sent an update even if you would have a leak each 1 minut it sent an update.

martijnvwezel, 2024-11-12:
Thank you for pointing that out  :) I will change to once

martijnvwezel, 2024-11-12:
Are you able already to connect with the WiFi?

Wiebrenk, 2024-11-13:
I have the logger running for over a month, really happy with it. Installation on the meter was a bit of a challenge but some tape helped. 

The connectivity loss does not create any problems in the energy reports. It does show up in the water usage graphs.
Logger has static ip, with DHCP lease of 1 day.

Logger: aioesphomeapi.connection
Source: runner.py:189
First occurred: November 12, 2024 at 11:12:39 (130 occurrences)
Last logged: 08:42:35

muino-water-meter-e5b248 @ 192.168.1.204: Connection error occurred: [Errno 104] Connection reset by peer
muino-water-meter-e5b248 @ 192.168.1.204: Connection error occurred: Ping response not received after 90.0 seconds

martijnvwezel, 2025-01-04:
can yous share a picture  how you taped it?

Wiebrenk, 2025-01-08:
Update; I manually updated the firmware from stock to V3.0.4 through the ESPhome in HA. This solved the connectivity problem. (I thought it automatically updated as part of the ESP home function)

Find below the taped sensor, I used the circle of the board to align the sensor to the indicator;

[image]

martijnvwezel, 2025-01-08:
https://watermeter.muino.nl/img/sensus_620.png

Not what watermeter you have exactly but try too look in the future if you have alligment issues to given picture above

closed 2025-01-08 by martijnvwezel