martijnvwezel/watermeter-esphome #58: Very inconsistent and negative water usage

JJ-White, 2025-01-03:
The meter is very inconsistent in detecting water usage. After the latest re-installation, it first reported no values, then way to low values, and has now been showing negative values for the day.

Some context:

I have the following water meter, which is on the supported list as far as I can see, but in the "use whatever method possible to attach the Muino Water Meter".

[image]

When I first installed the meter I used a zip-tie and added foam around the sensor area because I did not like the idea of the SMD components being in contact with the hard plastic of the meter. This setup then worked for a number of days, where at first it appeared to show accurate data, but in the coming days it was clear that water usage was not always registered. E.g. after a shower only 10L of usage was registered or none at all. Another issue here on Github mentioned light leakage might be an issue, even though the meter is behind a rack in a mostly dark basement I added a bag that completely blocks any light from coming in. This did not improve the performance.

[image]
[image]

Therefore, I set out to re-install the meter in a more permanent way, in case the loose zip-tie method shifted around in any way. For this second attempt I used double sided tape to secure the meter to the meter. To do this I removed the black foam from before. I think I eyeballed the alignment pretty well and stuck the water meter in place, it now has no possibility to move at all. 

[image]
[image]

Turns out, this installation performed even worse than before, as mentioned in the beginning. The first day there was no data at all, then some irregular usage and now negative usage. I think I have ruled out ambient light as an issue. I also used two different power supplies, to no avail. 

[image]

The only thing I can think of is that the Muino meter is confused by it's own light being reflected in the transparent plastic of the water meter, as this could be reduces with the black foam in my first installation. If this is the case I can design a small 3D print that separates the LEDs from the sensors and tries to block any reflections not coming from the dial.

I would like to hear if there are any other troubleshooting steps I can take or if there can be any explanation for this behavior?

martijnvwezel, 2025-01-04:
I have someone else with a simliar issue. I an just implementing a hardcoded calibration for this. I am currently writing a wiki how to deal with this type of inconsistenties and making sure that some with weird values or sensor that almost dont startup still work stable, downside of this hardcoded values is the moment alot of light is nearby and then its very dark the calibration will be wrong or can diviate. I will create a wiki, and in the following days I will try to finish and test the the hardcoded version. 

I was thinking about a seperate ring that seperate the light a bit, do you have a 3d printer?

martijnvwezel, 2025-01-04:
https://github.com/martijnvwezel/watermeter-esphome/wiki

i am starting to document things here, if you have any questions or things are not clear please tell, I will add it so everyone can use this as there guide.

4np, 2025-01-04:
Looking at your water meter, it has a reflective piece in top of the red dial arrow.

Correct me if I'm wrong, but the sensor works by detecting gaps in light emitted by a central led to determine the orientation of the rotating dial arrow. If the dial arrow _itself_ is reflecting light back because of a reflective piece in top of the red dial arrow, then the diodes come back with values the firmware is unable to make sense of.

I would think this is not related to ambient light as generally speaking a utility closet is pretty dark.

JJ-White, 2025-01-04:
@martijnvwezel I have a 3D printer and some experience 3D modelling. However, what I don't have is a lot of experience with ESPHome or Home Assistant, and I have not been able to find where I can enable debug mode? I only see the Speed mode switch and no other controls.

[image]

@4np I noticed those quirks too, but I see other supported meters also have this. I would imagine though that regardless of the shape and color of the dial, the sensor would detect a repeating patterns when the dial rotates. But if it only looks for peaks and valleys it might get in trouble. Lets see what the debug data says.

Unfortunately, I don't think any screw holes line up for my meter, but the double sided tape holds it just fine. Plus there should be no movement around this area of my house either.

martijnED, 2025-01-04:
Then you dont have the latest version running watermeter.muino.nl with chrome browser and you can reupload a new fresh version

martijnED, 2025-01-04:
I was started with a small packing for between the meter and sensor for optimal alligment so you can easily adapt it to your meter size

JJ-White, 2025-01-04:
When trying to connect using https://watermeter.muino.nl/ and selecting the right COM port, I get a "Failed to get manifest" error. The serial port does show up in device manager and the Chrome serial port list, so I do not suspect a driver to be missing. The LED on the meter is blinking away happily.

[image]

https://web.esphome.io/ and the https://esphome.github.io/esp-web-tools/ example do allow me to connect using Chrome and see the serial output.

4np, 2025-01-04:
I had the same issue updating the firmware (Chrome on macOS). As my sensor is working as expected I didn't bother and left the firmware as-is.

JJ-White, 2025-01-05:
I flashed the "[muino-water-meter-esp32c3.factory.bin](https://github.com/martijnvwezel/watermeter-esphome/releases)" file from the latest release using https://web.esphome.io/ and that worked to get me to v3.0.4 and HA now shows a debug switch and sensor. I will mount it back on the meter without any gasket in between and see what the debug output is.

JJ-White, 2025-01-05:
Not sure if I got the alignment better by accident, or something improved in the algorithm between v3.0.2 and v3.0.4, but it immediately calibrated after the 2 liters and speed mode now actually shows the values updating rapidly. Alignment was made easier because the 3D printed "Muino" cover popped off for me accidentally and I saw there was a circle outline of the sensor on the backside of the PCB. Perhaps the cover should also include such a feature? 

The debug output and plot doesn't mean much to me up front, but I've attached the HA export for the first 15-ish minutes and 12 liters here:

[history.csv](https://github.com/user-attachments/files/18311626/history.csv)
[image]

I will keep an eye on the values the coming days to see if it performs normally.

martijnED, 2025-01-06:
The graph looks good! So think the allignment is much better!

martijnED, 2025-01-06:
The esphome guys changes quite some things in the web portal, my feeling is that they changes more stuffand now the changed i also did dont work anymore

JJ-White, 2025-01-06:
@martijnED the history file is linked right above the screenshot.

So it seems to have broken again overnight, it's stuck on the same liters value and jumping up and down one step regardless of consumption. I've recorded two more history files, the first for slow consumption with the tap slightly open and the second with the tap wide open. 

There is a clear sequence in the plots, so the sensor is reading the dial, but the calibration seems way off to the point it's not detecting anything. The large downward spikes probably are not helping the auto calibration, without them the plot would be pretty clean... Is something over or underflowing in the detection code?

[image]
[image]

Files:
[history_fast_usage.csv](https://github.com/user-attachments/files/18317347/history_fast_usage.csv)
[history_slow_usage.csv](https://github.com/user-attachments/files/18317348/history_slow_usage.csv)

martijnvwezel, 2025-01-08:
I am really glad you gave soo much info. I see your issue. Somehow you have very weird noise in your signal. I have an idea how to solve it but need to implement it. 

I should implement a moving median sum filter to smooth out the signal and removing the noise 
and also have hardcode zero crossing line.

I will try to find time to implement it, not sure when tbo

martijnvwezel, 2025-01-10:
I have a spacer created: 
[Muino watermeter sensor standoff plate.zip](https://github.com/user-attachments/files/18375143/Muino.watermeter.sensor.standoff.plate.zip)

martijnvwezel, 2025-01-12:
I implmented in the latest version a way to make the calibration values static,  watch the wiki the comming days and will make a better explainaton their :
https://github.com/martijnvwezel/watermeter-esphome/pull/61

here the change explaination and how to calculate the values.

JJ-White, 2025-01-12:
Thanks for adding the manual calibration option! I just flashed it to the meter and I'll see what I can achieve in the coming days. I think with a good mount and a light covering bag the calibration values should be just fine as static values.

By the way, is there a way to remotely update the ESP over WiFi? Right now the only way I can do it is to unmount it and connect it to my desktop directly, which means I have to remount it every time. Perhaps I can do it with my laptop, but first try failed due to snap permissions on Linux not allowing access to the serial ports from Chromium.

martijnvwezel, 2025-01-12:
Yes you can update from remote, 2 ways just using the website and wait 24h for your home assistant to see the update and will ask for it, or using esphome add-on, you can copy my yaml and install remotely using esphome addon, if you have question feel free to ask :)

JJ-White, 2025-01-17:
@Wiebrenk I also don't get the notification for an update being available. I will try the ESPHome builder when a new version is out.

@martijnvwezel I designed and printed a 2.4mm thick gasket out of TPU that fits the Itron meter and your PCB, this makes alignment a lot easier and secure:

[image]
[image]
[image]

However, after mounting with this gasket the "bb" signal has gotten even weaker than it was before, making it difficult to find a mid point that is only crossed twice per rotation. I assume "bb" corresponds to one of the three light sensors, can you share which sensor corresponds to which physical location on the board? That might help with repositioning to improve the reading.

[image]
History; [history_gasket.csv](https://github.com/user-attachments/files/18453425/history_gasket.csv)

martijnED, 2025-01-17:
If I am correct, the signal b is too close to the edge:

[image]
Maybe moving it a tick to the left would help?

The amount of signal should not matter too much. 

Cannot find the issue about the update frequency how this exactly work yet.

martijnED, 2025-01-17:
I can help allignment with a different 3D printed form, but need some help giving me dimentions:

[image]

[image]

martijnvwezel, 2025-01-17:
maybe we can create something for those, 

[image]
small bums so it clicks into the watermeter:

[image]

JJ-White, 2025-01-18:
I spent all of last night prototyping a proper bracket that uses the clipping points on the Itron to have no more guesswork in the mount. I think I succeeded:

[image]
[image]
[image]

This is the view through the window while the dial is turning:
https://github.com/user-attachments/assets/e58fa427-3529-44ac-a495-61830c688722

The signal is a lot better now, especially for the "bb" channel, and should be repeatable when remounting. The "bb" channel is still lower than the others for some reason. I put in manual calibration values of A=380, B=230, C=310.
[image]

I will put the bracket model on Printables if there are no more changes.

JJ-White, 2025-01-18:
Now that the signal is pretty good, I do notice an inconsistency in the detection algorithm. The milliliter sensor does not consistently jump up in the 167mL steps when I leave the tap running at a slow but constant speed. Instead, it gets "stuck" on the half liter marks after which it jumps two steps in rapid succession. This plot shows the behavior:

[image]
History file: [new_mount_slow_constant_usage.csv](https://github.com/user-attachments/files/18463906/new_mount_slow_constant_usage.csv)

Not a major issue as accurate liter usage is most important, but for a sensor that advertises mL support it should report them consistently at least.

martijnED, 2025-01-18:
It works in steps of 1/6 because of using less complex math. It can be smaller mililiter precision but bit more noise. Did you enable speedmodus? Maybe thw edge going up is too steep on your meter, so some steps are invisible.

martijnED, 2025-01-18:
If you tapp 100ml do you see it in the usaged?

thibautsim, 2025-01-19:
Hi, 
I'm having a similar issue (my sensor is counting less water that what is really used) and I'm trying fo follow along but it is quite technical for me 😄 

So far : 
I have managed to update my sensor with latest firmware. 
I have eventually managed 1) to find how to deal with the debug mode and retrieve data and 2) how to use the CSV visualiser. 

[image]

From there I tried to apply what I read here : 

[image]

What I did is to analyse each sinus curve individually to find out the min and max values. In my case for example for the state.aa it was min = 341 and max = 647.  

[image]

Used Excel to help me determine the offset for each curve

[image]

And put those value in HA 

[image]

But I must have done something wrong because now, the situation is even worse than when I started. The meter is metering very little water. 

@martijnvwezel Can you help me ? What did I do wrong ? 
Here is the csv file if needed : [history (3).csv](https://github.com/user-attachments/files/18469107/history.3.csv)

Thx !

JJ-White, 2025-01-19:
@thibautsim the way I interpreted the manual calibration is just to pick the center point of each signal, the manual values you have appear to be much too low. What happens if you put in the mid point between the min and max values for each channel?

thibautsim, 2025-01-19:
@JJ-White Thanks for your reply. It's what I initally understood to. So I've modified the formula to determine it as follow : (max-min)/2 + min. I have updated the values in HA with the results. 

[image]

I've made a test while my children were taking their showers. Real water usage : 132 liters. Measured by Muino : 49 liters. 

To be on the safe side I've taken a debug file during that period too. Here it is below (printscreen + file). It seems this time there are moments where it goes below 0. 

[image]

[history (4).csv](https://github.com/user-attachments/files/18469719/history.4.csv)

Any ideas ?

JJ-White, 2025-01-19:
Just guessing here, but looks like the regular "dip" in the middle of the signal peak for channels "aa" and "bb" are crossing the manual calibration point. Probably something to do with the shape or material of the dial on your meter. Although, the "cc" channel doesn't have it so it could be mount related?

What happens if you put the manual calibration value below these dips, like 450 for A and 220 for B? Then it should only register two crossings for every rotation instead of four.

@martijnvwezel I uploaded the model to Printables here: https://www.printables.com/model/1155682-itron-bracket-for-muino-water-meter-reader

JJ-White, 2025-01-19:
Regarding the accuracy when taking 100ml of water this is my initial finding with debug and speed mode disabled:

- Reading before 408 500 ml
- Take 100 ml in a measuring cup
- Wait 2 minutes and reading shows 408 666 ml
- Take 100 ml in a measuring cup
- Wait 2 minutes and reading still shows 408 666 ml
- Take 1 000 ml in a measuring cup
- Wait 2 minutes and reading shows 408 833 ml
- Wait 5 minutes and some random usage in between, sensor still reads 408 833 ml

So far, not good results in normal mode...

JJ-White, 2025-01-25:
I did another check for the meter's accuracy, this time by photographing the Itron value before and after filling a bath and other morning activities. The Muino meter was in normal mode, with the same manual calibration values as before. I validated these values with another debug export yesterday and they still align with the center points of the signals. 

During this 1.5h period the following values were reported by both meters:

- Itron: 70L
- Muino: 33L

[image]

I can live with inaccurate mL readings, but the overall liters value should definitely be accurate otherwise there is no point. Looking at the signals, the data should be plenty accurate to determine an accurate liter value, even a 1/3rd liter value should be bullet proof, so where is this going wrong?

thibautsim, 2025-01-26:
Hi @martijnvwezel 

Any chance you can help @JJ-White and myself regarding our last posts ? 

Thanks !
Thibaut

martijnvwezel, 2025-01-26:
Do you have the csv output for analysis @JJ-White ?

martijnvwezel, 2025-01-26:
Whas a bit busy yesterday. I want to understand the issue. I myself playing around with filtering the signal more. meters with metalic plating is a bit noisy and needs to be addressed

martijnvwezel, 2025-01-26:
@JJ-White  tonight I have some spare time to look at your data and do some analysis. I am pretty sure the manual calibration is the issue. I am also working on a small filter but need some more debugging

JJ-White, 2025-01-26:
@martijnvwezel I don't, because the meter was in "normal mode" because you mention somewhere that should be more accurate and I wanted to give it the best opportunity.

I will repeat on the mext shower/bath opportunity and enable debug mode and speed mode.

martijnvwezel, 2025-01-26:
I missed quite some messages in this thread.  Sorry for that I see you also gave me some debug files 

[image]

I know they are changing the ADC in the ESP, so my feeling is the substraction of the dark light is maybe an issue. Here the values goes below zero what should not be the case at all 
[image]

History file: [new_mount_slow_constant_usage.csv](https://github.com/user-attachments/files/18463906/new_mount_slow_constant_usage.csv)

ignore the savera or the sum thingy, but the 
[image]

### 120 degrees
You comment about the half liters, Feel really dump now. That I didn't see before your meter dail is not 50% dark like mine. Yours is 120 degreese only. So if you look at the graph below the algorithm works just fine, but because the peak is more shallow (shorter) you have 20% quicker stepsthats what you mean that it jumps with yours more 0.2L faster in the brighter time period. 

[image]

As the origin of the algorithm is sinus waveform based it will look for the algorithm like this below.
I have to think a bit how to do it, but I could change the formula a bit to match your  meter. Never though about this issue. But future me problem. First focus on the missing liters what is a huge issue! Hope the guys from home assistant will not break the adc code, they are they it to the new way of esp-idf :/

[image]

JJ-White, 2025-01-26:
Thanks for having a look! I just recorded a debug log while filling a bath and noticed some other oddities.

The overall usage seems to jump down frequently while water is being consumed at this rate:

[image]

Probably related, the phase plot looks odd. Sometimes is follows the steps nicely but often it jumps between 0 and 5 like it doesn't know what to do. 

[image]

Looking at the signal quality the peaks are still very well defined, albeit plagued by the regular downward spikes. Speaking of those, they appear in a bunch like clockwork every 60 seconds regardless of usage. This could indicate there is some process that happens every minute that interferes (this recording was in debug mode, so WiFi should be active constantly). Or perhaps there are two processes that go in and out of phase with each other to interfere periodically. I'm not familiar enough to know what stuff is going on in the ESP that could cause this, of if the analog meter can have any influence on the signal.

[image]

Here is the history file in case you need more debug info: [bath_time.csv](https://github.com/user-attachments/files/18551421/bath_time.csv)

During this test the usage according to the Itron meter was 94L and the Muino reported 25L was used.

Let me know if there's anything else I can test or do to help bring improvements!

JJ-White, 2025-02-06:
@martijnvwezel any timeline on when you think you can have a fix implemented, at least for the liter count so the sensor is usable?

thibautsim, 2025-02-07:
Indeed, it would be nice to have an software update so we can use the meter 👍

martijnED, 2025-02-10:
Timeline was this weekend, but i am sick i want to take a fed days off to work on it but first wane be better

JJ-White, 2025-02-23:
Hey @martijnED, any update on this?

martijnvwezel, 2025-02-23:
Hey @JJ-White , I was sick for quite some time. However, since yesterday I am testing the smooth filter. On my meter I do not have the same issue as you. However, I don't see it got worse or incorrect. Would like you to try the version in develop. Are you able to install it yourself? Try the commands from production script when you checkout develop. You can also copy all code from the muino yaml file and then use esphome in you ha setup to install it.

martijnvwezel, 2025-02-23:
Its this pull request: https://github.com/martijnvwezel/watermeter-esphome/pull/64

martijnvwezel, 2025-02-24:
I tested the code seems to work!

martijnvwezel, 2025-02-24:
I already though about how to make it for your meter dail working, Need to change the algorithm a bit for that, so the way the liters are calculated is more in sync. Will try to work on it somewhere this week too

JJ-White, 2025-02-26:
Thanks for the update, I used ESPHomeBuilder to install the `develop` branch. At least, I think I did because I cannot see the commit hash or branch anywhere in the log or interface.

[image]
Update log: [logs_muino-water-meter-e5f28c_run(2).txt](https://github.com/user-attachments/files/18992792/logs_muino-water-meter-e5f28c_run.2.txt)

The debug log after the update still shows the spikes present and incorrectly counts the water usage. I am not sure I am running your development version, because I see no difference compared to previous release (I made a debug log right before updating to compare).

[image]
History file: [history.csv](https://github.com/user-attachments/files/18992854/history.csv)

Please let me know if I'm doing the update with ESPHomeBuilder incorrectly.

martijnvwezel, 2025-02-26:
its in the master branch

martijnvwezel, 2025-02-26:
`The spikes once in the while when its doing not much is not a problem

martijnvwezel, 2025-02-26:
you can also copy the whole yaml  in this repo and then I can help you tune it

martijnvwezel, 2025-03-02:
I am currently testing the correct way of increasing the liters for your meter. So you need to configure it in epshome device part using a button

martijnvwezel, 2025-03-02:
Also working on a improved version of the algorithm that updates the max only on specific moments, improving the stability for your meter

thibautsim, 2025-03-03:
Hi @martijnvwezel 
I had a similar issue than Juriaan as my meter is counting less than the real usage of whatever (see my previous posts).  
I don’t understand what I should do. Is there already a new version I can try ? I only have basic technical skill so if you could explain it in a user-friendly way, that would be great. 
Thanks.

martijnED, 2025-03-03:
Yeah basically there are 3 issues. 
* auto calibrations - doesn't currently work, because it works well of half of the disk is covert but yours has now 120 degrees [in latest version]
* lot more noise from the metal reflection - possible solution is now in master, [in latest version]
* liters don't increase correctly but the liters are correct. I am implementing a way that can deal with 120 degree disks so that the milititers additions is corrected for your water meter [not finished]

you can update the watermeter using three ways one is using automatically update from the home assistant. Not sure if that now works compleetly, they changed many things. One is using the watermeter.muino.nl website, or just copy the yaml from this repo in you home assistant esphome addon. you can ask openai to assist you with that part or just this chat.

martijnvwezel, 2025-03-21:
Juriaan can you share me your telegram or whats app please?

martijnED, 2025-04-01:
Martijnvwezel @ muino.nl

thibautsim, 2025-04-09:
> Yeah basically there are 3 issues.
>
> * auto calibrations - doesn't currently work, because it works well of half of the disk is covert but yours has now 120 degrees [in latest version]
> * lot more [...]

Hi @martijnED 
I hadn't had time to look at this properly. 
My version of the meter is : Firmware: v4.0.4 (ESPHome 2024.10.3)
Should I update to your v4.0.5 🌈 Big bug fix manual button true ? 
Will this fix the issue ?

martijnvwezel, 2025-11-28:
Updated

closed 2025-11-28 by martijnvwezel