martijnvwezel/watermeter-esphome #60: New Diehl Watermeter incompatible?

VDRainer, 2025-01-12:
# Problem/Motivation

After a new water meter was installed at my home, i get absolute no values anymore.
The old Watermeter worked for over a year (not 100% acurate) without problems.

Now i just want to know if the spinning disk in the new meter is incompatible and i can stop tinkering with muino, or if there's a chance to get it working.

Old Meter worked:
[image]

New Meter no values at all:
[image]

I tried version 3.0.0 and 3.0.4, but simply no chance to get a value.

A few debug_json values i copied with 3.0.4  look like some data changes, but i have no clue how to deal with it. :cry: 
```
{"liters":1,"phase":5,"last_reported_liters":1,"last_reported_liters__mili":1833,"aa":95,"bb":115,"cc":70,"max_a":255,"max_b":207,"max_c":243,"min_a":10,"min_b":16,"min_c":12,"upper_bound":1500,"lower_bound":5}
{"liters":0,"phase":5,"last_reported_liters":0,"last_reported_liters__mili":833,"aa":177,"bb":188,"cc":93,"max_a":506,"max_b":318,"max_c":301,"min_a":6,"min_b":14,"min_c":10,"upper_bound":1500,"lower_bound":5}
{"liters":0,"phase":5,"last_reported_liters":0,"last_reported_liters__mili":833,"aa":122,"bb":194,"cc":94,"max_a":506,"max_b":318,"max_c":301,"min_a":6,"min_b":14,"min_c":10,"upper_bound":1500,"lower_bound":5}
```

Thanks in advance for reading this!

martijnvwezel, 2025-01-12:
Latest version 4.0.0 has a offset value that you can set statically: https://github.com/martijnvwezel/watermeter-esphome/pull/61

The moment you find a better zero -crossing line:

[image]

you can update the parameters and it uses that  and will not change anymore

martijnvwezel, 2025-01-12:
in your situation I need a bit more info if you want my help see the wiki: https://github.com/martijnvwezel/watermeter-esphome/wiki

martijnvwezel, 2025-01-12:
btw first 2 liters are never shown :)

martijnvwezel, 2025-01-12:
With https://watermeter.muino.nl/ you can update your meter to the latest or wait for your HA to tell you this. But after that you can make static values as shown above

VDRainer, 2025-01-13:
Hi @martijnvwezel , thanks for your fast response and your tips. :+1: 

I must say, i was a little bit confused about your instructions.

What i found out in other issues with some example data:
- the sensor.debug_json has to be recorded in the (influx) database
- export some timeframe of that data to a csv file
- import that csv file in your CSV Visualizer

I don't use influxdb and dont want to install it just for debugging, so i tinkered with another solution to get the data.
1. Create a [file notification](https://www.home-assistant.io/integrations/file#notifications)
2. Create an automation that saves the state of the debug_json sensor to that file.
```
automation:
  - alias: muino_debug_json_logger
    mode: queued
    trigger:
      - platform: state
        entity_id: sensor.debug_json
        variables:
          the_json: >
            {{ (trigger.to_state.state|to_json(ensure_ascii=True)).replace("\\", "\"") }}
    action:
      - service: notify.send_message
        data:
          # entity_id,state,last_changed
          message: "sensor.debug_json,\
                    {{ the_json }},\
                    {{ now().strftime('%Y-%m-%d %H:%M:%S') }}"
        target:
          entity_id: notify.muino_debug_json
```
Now, the json is recorded in a file and all you have to do is copy that file (or parts of it) to your local computer,
add the csv header `entity_id,state,last_changed` in the first line,
and then import it in https://watermeter.muino.nl/data_visualiser.html

If i look at values i recorded, i think i have to experiment with the mount for the sensoror on my meter?
Maybe you can say something about it?
[muino_debug_json.csv](https://github.com/user-attachments/files/18399939/muino_debug_json.csv)

VDRainer, 2025-01-16:
Created a new stable mount.
Experimented with 4.0.4 Manual Calibration, Speed mode on/off, ...
Still no values. :sob: 
[image]
[muino_debug_json.csv](https://github.com/user-attachments/files/18435495/muino_debug_json.csv)

martijnvwezel, 2025-01-18:
Sorry it took me a bit of time to investigate

Those values you used for the offset are quite wrong tbo.
Can you try the suggested:
``` python
Offset_a = 100 
Offset_b = 130
Offset_c = 75

```

martijnvwezel, 2025-01-18:
The min max algorithm works for many meters but not all. However the reason why you wanne have min max is that the algorithm has to find the middel point of the signal. so it has a reference where it is in the signal. Zero crossing is the line where the sinus normally cross the 0 of the vertical axis. However with this signal thats not the case  therefor the above values telling where the 0 should be essentially. 

[image]

VDRainer, 2025-01-18:
Thanks for the response.

I took the values from the csv with your [formula](https://github.com/martijnvwezel/watermeter-esphome/pull/61) 
`manual_offset_a/b/c = (max_a/b/c - min_a/b/c)/2`
Your values ​​are determined by `manual_offset_a/b/c = (max_a/b/c + min_a/b/c)/2`

Even with your values ​​I don't get any liters. 

[image]

[muino_debug_json.csv](https://github.com/user-attachments/files/18464778/muino_debug_json.csv)

martijnvwezel, 2025-01-26:
The values I took are basically from looking at your signal and finding the max of it not looking to the debug values. The max goes compleetly wrong in your case.

martijnED, 2025-01-29:
I had some issues with my website and many questions will plan some days off to work on this project more. I want to support as many meters as possible and reliable but will take bit more testing.

VDRainer, 2025-01-29:
Hi Martijn, thanks for your responses.

I'm sorry to say this, but I have switched to 'AI on the edge' in the meantime.
The fact that I have not received any values at all ​​leads me to suspect that the disc in my new watermeter is incompatible.

I can't promise, but maybe I'll test a newer version in the future.

Maybe you want to use my [instructions](https://github.com/martijnvwezel/watermeter-esphome/issues/60#issuecomment-2587724244) to record the debug json in your wiki?

> 4. due lack of time ask openai temporarly to help!

Closing this for now.
Thanks again for your effort. 👍

pawel-b, 2025-01-29:
@VDRainer, can you share what are the differences between Muino and AI on the edge when it comes to precision and accurracy? I've tested briefly AI on the edge and found that it will report the final correct water usage eventually, but at the same time the intermediate values can be incorrect.

VDRainer, 2025-01-29:
> I've tested briefly AI on the edge and found that it will report the final correct water usage eventually, but at the same time the intermediate values can be incorrect.

Yepp, the intermediate values can bi incorrect, but at the end, i have accurate values.
At the moment the most important thing for me was to get values ​​at all. 😄

martijnvwezel, 2025-11-28:
Couldnt find your toppic because this was closes :/

martijnED, 2025-11-30:
Can you check the software version using watermeter.muino.nl you can uodate the meter.

patbro, 2025-12-07:
I updated the watermeter to version 4.0.9, I set the calibration values again, turned on the manual calibration toggle, and even restarted the watermeter after a while.

The `liters` entity still reports 0 L. Though, the `flow rate current` reports some data. The `debug json` entity remains 'unknown' even though I toggle the debug mode, so that makes it hard to test.

The entity `spindel` is new in this version, what does it do? The docs don't mention it.

[image]

[image]

martijnED, 2026-01-12:
How did you update the esp?
Can you try to use it using the esphome tool in home assistant itself, and copy the full yaml file?
Do you need help with that?

I want to understand exactly what goes wrong with your setup you have weird issues :/

martijnvwezel, 2026-01-18:
make sure your manual offset is refilled in after update :)

patbro, 2026-01-20:
Hereby, a quite rough, JSON file with the debug data: [muino2.csv](https://github.com/user-attachments/files/24723005/muino2.csv) and another one with a shorter timespan: [muino2-1.csv](https://github.com/user-attachments/files/24749778/muino2-1.csv)

Again, the flow rate current seems correct. The current consumption reports negative as well as positive values. Seen the debug data, the watermeter seems to be aligned correctly.

Just now, I reprogrammed the Muino via the CLI with esphome, the log returns the following from time to time:
`[W][light_level:548]: Signal too weak. Possible causes: too bright environment, LED failure, or sensor misalignment. Ignoring measurement`
The watermeter is mounted in a dark electrical closet.

martijnED, 2026-02-19:
@patbro Hey do you mind to have a quick call about this. I would like to impleemtn a fix for your meter

arjanmels, 2026-03-08:
I have finished my updates, see pull request: #85. However, I don't think it handles this case properly yet: there seem to be 2 peaks for each sensor per rotation: this means 12 phases instead of 6 the (and a lot of non-unique phases).
The code could be extended to handle this, but does require quite a bit of work (especially on the yaml sensor definition as you cannot use arrays there) and there is no elegant way to handle both 6 and 12 phases in the UI. (Would need to include all patterns and liters per phase and then only use 6 for most meters).