martijnvwezel/watermeter-esphome #65: Debugging steps, how to do

Joska59, 2025-03-30:
Hi Martijn

I also observe a certain drain of the measured values, so that the recording no longer works after a few days/hours. Negative values or values for water consumption are measured. Now I have read through various texts here in Github and am not getting smart enough. The Water-meter can start a debug mode with recording of a lot of data at the push of a button, but it is not clear to me where the data ends up. I see this data in HA's log, but the way I should proceed with it remains unclear to me due to the lack of InfluxDB or the connection to it. Can I not save the debug data? If I cannot simply save it, but can read it, is manual hardcoding somehow possible? Another user wrote that the formula you mentioned does not seem to provide any useful values. 

Regards, Jörg

Joska59, 2025-04-01:
I tried to get some debug Infos.

{"liters":796,"phase":2,"last_reported_liters":796,"last_reported_liters__mili":796333,"aa":139,"bb":130,"cc":133,"max_a":5,"max_b":5,"max_c":5,"min_a":5,"min_b":5,"min_c":5,"upper_bound":1500,"lower_bound":5}

In this case, depending on the two formed you have written, (min_a+max_a)/2=5, same with all other values. Can you describe it better for older, not nativ-techie-people? 😉

Joska59, 2025-04-01:
Now, after a reset, It:

{"liters":10,"phase":2,"last_reported_liters":10,"last_reported_liters__mili":10333,"aa":71,"bb":118,"cc":352,"max_a":176,"max_b":127,"max_c":345,"min_a":62,"min_b":101,"min_c":121,"upper_bound":1500,"lower_bound":5}

So, manual calibration offsets should be

a=(176-62)/2=57
b=(127-101)/2=13
c=(345-121)/2=112

or

a=(176+62)/2=119
b=(127+101)/2=114
c=(345+121)/2=233

Depending of the two different formed, you have posted. I suppose, the second one is correct.

Can you describe, how to restart the device, using an automation? The inbound properties seems not to be able to do?

Merc3nary, 2026-09-13:
Same here, works for a couple of days and then starts to degrade with negative measurements or just constant values.
Would be nice if this could be fixed.